arduino/arduino-builder #317: Hooks for early upload prep & cleanup

PaulStoffregen, 2019-04-03:
I'd like to request special hooks to run recipes intended to minimize the wait time for uploading.

A hook similar to recipe.hooks.sketch.postbuild.NUMBER.pattern is needed, but only run if the build is for Upload.  If the user clicked Verify, this recipe should not be run.  The idea is to begin the process of a board rebooting as early as possible (but after the user's sketch it known to compile without errors), to allow time for USB enumeration and driver initialization while the build process continues.

Another build error hook would probably also be needed.  If building any library, the core lib, or linking fails, the error hook would run a recipe which quits bootloader mode and causes the selected board to reboot to whatever code was previously programmed.

For the common case of relatively small sketches on native USB boards like MKR & Micro (and Teensy), the time for USB enumeration is often the a large chunk total time required to upload.  If we could start that early in the build process, users could have a much quickly total upload experience.

facchinm, 2019-04-18:
Hi Paul,
I think the main issue here is related to the 1200bps touch (not relevant on the Teensy). After the touch, the IDE waits for the serial port to disappear (without enforcing any check on vid/pid); if the board is already in bootloader it takes much longer to upload since it waits until the timeout is reached.
So the hooks are not super useful on our side (also, MKR have no software way to safely exit bootloader since 1200bps touch calls `banzai()` and destroys the user sketch ).
Adding the hook alone won't hurt though. Do you mind opening a PR?

PaulStoffregen, 2019-04-19:
Must confess, I haven't worked with the builder code, nor with Google Go language... so probably no PR from me anytime soon.  If I had to do this one my own, I'd just put an ugly kludge into the Java code.

facchinm, 2019-04-19:
No worries, I will remember this request as soon as we touch the builder (or the cli) again :wink: