arduino/arduino-builder #326: Lock using static libraries *.a

luluv1, 2019-06-24:
Hello,

I would like to link a static library into my Arduino project. I have tested it for SAMD and AVR respectively ( Arduino Uno et Arduino Zero)

I have already read this topics:
https://arduino.stackexchange.com/questions/57891/how-to-use-a-precompiled-library-in-a-project-with-arduino-ide
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
https://github.com/arduino/Arduino/issues/4336

I follow the instruction in order to add static library, then I can see it in the Contributed library (attached file)

Here is the source code of my compiled library :
`
int foo(void) {
  return 1;
}`

Then, I create a library using ar tool.

I create a repo called foo which include
* library.properties
* and a repo src as below:
-/src
---foo.h
---/{architecture_mcu_name}
-----libfoo.a

```
author=lululuv1
maintainer=foo <foo@gmail.com>                                                                          
sentence=                                                                                                
paragraph=                                                                                              
category=Uncategorized                                                                                  
url=                                                                                                    
architectures=*                                                                                          
precompiled=true                                                                                        
ldflags=-lfoo`
```

My foo.h contain the function prototype : int foo(void);

FInally I create the ino project, include the library.... 

```
#include <foo.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  int foo_value = foo();
  Serial.print("FOOFOOFOOFOOF");
}

```

Link with the library is done because if I want to link a libfoo.a from a dismatch architecture I got an error.

Hence, there is no compilation error, no link error but when i call the function foo(); I am lock, the print FOOFOOFOOFOO is never done.

Any idea ?

[image]
[image]

Thank you
PS: this is a simple example in order to reproduce the problem, can be done with avr-gcc or arm-none-eabi-gcc

kgoveas, 2019-06-24:
Hi,

Check out the readme in https://github.com/BoschSensortec/BSEC-Arduino-library

facchinm, 2019-06-25:
Hi @luluv1 ,
I can't see what could be wrong with your setup; could you try a simple hack to understand if the problem is with the lib you are linking?
1 - remove `precompiled=true` for your library.properties and add `dot_a_linkage=true`
2 - add a C/C++ file with foo() implementation 
3 - compile the example once

This will produce a `.a` archive of the library that you can copy back to the lib folder; after undoing the changes (remove implementation file and readding precompiled flag) should result in the same outcome.

Let me know if it works or there's any improvement!

luluv1, 2019-06-26:
Thank you @facchinm ,

I did all the steps. Using a foo.cpp, it creates a foo.a in the tmp folder. 
I try to use this one, compilation is ok. I can upload the code but there is no improvement... I did this test using Arduino M0 Pro

I attach the library created using dot_a_linkage -> [libfoo.zip](https://github.com/arduino/arduino-builder/files/3328771/libfoo.zip)

Thank you very much for your help,

facchinm, 2019-06-26:
Hi @luluv1 ,
I tested your .a and it works just fine in my setup (SAMD core 1.8.1). I'm attaching the "full" library so you can test too in your setup
[libfoo.zip](https://github.com/arduino/arduino-builder/files/3328812/libfoo.zip)
Let me know if it works for you!

luluv1, 2019-07-09:
Hi @facchinm ,

You are right, it works fine. 
When arduino create the foo library, it use the same compilation flag... 

My mistake when I create the foo library on my own. 
--->  I do not use the `-mthumb` flag unlike the arduino compilation

Also, I find another interesting point. (Which is maybe obvious for you)
In my library (not the foo library but a real one ;-)  there is an (useless) empty `main` function. I guess that Arduino consider this main as the entry point. Hence, nothing happen...

Thank you for your help.

closed 2019-07-09 by luluv1