arduino/arduino-builder #329: Static library refuses to be linked properly

PvdBerg1998, 2019-07-17:
I'm trying to use a precompiled library (written in Rust, detail here https://rust-embedded.github.io/book/interoperability/rust-with-c.html). This should all be set up properly and generate a correct static library.

I'm using a Maple Mini, so I compile my Rust code for `thumbv7m-none-eabi`. This generates a `lib<name>.a` file which I placed inside my library folder:
```
src
  cortex-m3
    lib<name>.a
  <name>.h
library.properties
```

The `library.properties` file includes the flags:
```
precompiled=true
ldflags=-l<name>
```

However, this doesn't even add the set linker flag to the verbose compiler output I see in the Arduino IDE. After some googling I found that the `platform.txt` may not include this flag, so I added it to `Arduino/hardware/Arduino_STM32/STM32F1/platform.txt`:
```
# snip
# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags="-L{build.variant.path}/ld"
compiler.cpp.extra_flags= 
compiler.S.extra_flags=
compiler.ar.extra_flags=
compiler.elf2hex.extra_flags=
compiler.libraries.ldflags=  # NEW
# snip
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -lm {compiler.libraries.ldflags} -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group {object_files} "{archive_file_path}" -Wl,--end-group
```
Notice the `{compiler.libraries.ldflags}`. This does pass the flag correctly.

The structure of the library is exactly the same as the library https://github.com/BoschSensortec/BSEC-Arduino-library.

Result of this adventure:
```
C:\Users\x\AppData\Local\Temp\arduino_build_200877\sketch\rust_lib_test.ino.cpp.o: In function `loop()':

C:\Users\x\Documents\Arduino\sketches\rust_lib_test/rust_lib_test.ino:10: warning: undefined reference to `_hello_rust'
```

I'm getting extremely frustrated by this issue, I hope someone can help me out.

kgoveas, 2019-07-18:
Hi,

At the variable in the archive group of the combine recipe. -Wl,--start-group {object_files} "{archive_file_path}" {compiler.libraries.ldflags} -Wl,--end-group

PvdBerg1998, 2019-07-18:
For anyone stumbling upon this issue, the final section is:
```
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -lm {compiler.libraries.ldflags} -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group {object_files} "{archive_file_path}" {compiler.libraries.ldflags} -Wl,--end-group
```
@kgoveas Thank you so much, it's working now. ❤️

closed 2019-07-18 by PvdBerg1998