arduino/arduino-builder #336: Unable to use arduino-builder released with 1.8.10

fpistm, 2019-09-19:
Hi @facchinm ,

arduino-builder  **v1.4.7** is not usable. Using 1.8.10 which include this I'm could not make any build.

Using https://github.com/arduino/arduino-builder/wiki/Doing-continuous-integration-with-arduino-builder example:
```Console
./arduino-builder -hardware ./hardware -tools ./hardware/tools/avr -tools ./tools-builder -libraries ./libraries -fqbn arduino:avr:uno /local/data/work/IDE/arduino/arduino-1.8.10/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino
stat : no such file or directory
```
with arduino-builder **v1.4.4** including in 1.8.9 this works fine.

Same issue using `-build-options-file` generated with Arduino IDE. Failed with 1.4.7, passed with 1.4.4.

Finally using our CI build using 1.8.10, error is 
`fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory`

See logs here:
https://travis-ci.com/stm32duino/Arduino_Core_STM32/jobs/236988204

I've reproduced on windows 10 and Linux (Ubuntu 16.04).

Best regards.

facchinm, 2019-09-19:
Hi Frederic,
the builder as it is now (after merging https://github.com/arduino/arduino-builder/pull/319 ) is no more the go-to tool for CI jobs.
The java IDE populates the paths as needed but we gave no extra care about different usage (like CI) since we are moving everything (also the builder sources) to https://github.com/arduino/arduino-cli .
Sorry for not giving any official notice about this.
I'm leaving the issue open since it's probably just a matter of calling `filepath.Abs` here and there but I strongly advise you to switch to the cli (which makes other CI mundane tasks very easy)

fpistm, 2019-09-19:
Thanks Martino for the quick answer.
I will have a look to arduino-cli.
Anyway, I think the arduino-builder v1.4.7 is no more inline with its documentations and help.

cmaglie, 2019-09-20:
@fpistm 
may you try this build: https://github.com/arduino/arduino-builder/pull/337#issuecomment-533614339 to see if this fixes the problem?

fpistm, 2019-09-22:
Hi @cmaglie 
I've tested under windows and this works fine now.
Thanks a lot for the quick fix.

PR #337 (2019-09-23): Fix buildpath regression
WIP: do not merge before https://github.com/arduino/arduino-cli/pull/418

There were actually two bugs:

- the command line parsing check, to see if the `build-path` parameter has been passed, should be done against the empty string (`""`) and not against `nil`
- the old `ctx.ToolsDirs` parameter has been replaced by `ctx.BuiltInToolsDirs` but this has not been reflected in the builder main.go

Not related with this PR but required to compile:
- the function Unquote has been moved from arduino-cli to this repository

Should fix #336
closed 2019-09-23 by cmaglie