arduino/arduino-builder #339: Building from local and modified sources ?

d-a-v, 2019-09-22:
Hello,

I can't find in documentation how to compile arduino-builder from my local sources.

I can `git clone` then `go build` and it works.

However when I modify something in `src/github.com/arduino/arduino-cli/` I can never get my sources properly compiled.

Tweaking by pushing updates in an arduino-cli fork and updating github refs gives some results - my sources are compiled - but in the end this process fails to get to the final arduino-builder executable because of unmatching git urls.

Thank you for your help.

facchinm, 2019-09-23:
@cmaglie any hint on how to work with this flow when using `go.mod` ?

cmaglie, 2019-09-23:
it's very simple, change your `go.mod` by adding a replace line, like this:

```
module github.com/arduino/arduino-builder

go 1.12

replace github.com/arduino/arduino-cli => ../arduino-cli

require (
	github.com/arduino/arduino-cli v0.0.0-20190920124130-b6774033151c
	github.com/arduino/go-paths-helper v1.0.1
	github.com/arduino/go-properties-orderedmap v0.0.0-20190828172252-05018b28ff6c
	github.com/go-errors/errors v1.0.1
	github.com/juju/errors v0.0.0-20190207033735-e65537c515d7 // indirect
	github.com/sirupsen/logrus v1.4.2
)
```

once saved the go tooling will ignore the dependecy written in the `require` section and will use the relative path to your local copy specified in the `replace` line.

d-a-v, 2019-09-24:
Thanks !
This should be documented in CONTRIBUTING.md

To build with latest `arduino-cli`, I had to do (in `arduno-cli`):
```
git log legacy/builder/gohasissues/go_has_issues.go
git diff 25cb6789ae73d9f69393d2afbd0699f5bcb041c8..d94fe6c2fa894cfe67f6aa8866261739afbd6884 legacy/builder/gohasissues/go_has_issues.go | patch -p1 -R
```
(because of:) 
```
./main.go:289:20: undefined: gohasissues.Unquote
./main.go:303:28: undefined: gohasissues.Unquote
./main.go:323:34: undefined: gohasissues.Unquote
./main.go:342:34: undefined: gohasissues.Unquote
./main.go:426:13: undefined: gohasissues.Unquote
```

With the arduino IDE and https://github.com/arduino/arduino-cli/pull/421 I can now build external arduino projects that were designed to be managed by PIO only.

cmaglie, 2019-09-24:
I see, well we're going to merge https://github.com/arduino/arduino-cli/pull/421 so this will not be a problem anymore.

cmaglie, 2019-10-01:
There should not be anymore build errors in the master branch due to dependency on `arduino-cili`.

closed 2019-10-01 by cmaglie