arduino/arduino-builder #341: Regression in 1.8.10: Segfault in arduino-builder with -dump-prefs if no sketch is specified

obra, 2019-09-24:
[tl;dr: In 1.8.10, something changed in arduino-builder such that -dump-prefs no longer works without a sketch and -build-path specified. On top of that, there's a segfault when trying to abspath....something if a -build-path is provided, but no sketch is provided.]

Hi folks,

We've been seeing some weird behavior with 1.8.10. It manifested at first as 
```
   fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory
```

As I dug in, I discovered that the deeper issue was that arduino-builder was segfaulting when I ran `arduino-builder -dump-prefs` with a -build-path without a sketch:

````
$ /usr/local/arduino-1.8.10/arduino-builder -hardware /usr/local/arduino-1.8.10/hardware  -tools /usr/local/arduino-1.8.10/hardware/tools -fqbn arduino:avr:leonardo   -build-path /tmp/  -dump-prefs 
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x5346a7]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).Abs(0x0, 0xc000021400, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/go-paths-helper/paths.go:139 +0x27
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xfa4300, 0xc000021400, 0xfa4300, 0x40c3c8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_setup.go:67 +0x34e
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000021400, 0xc00018fe00, 0x2, 0x2, 0x42b001, 0x8, 0xc000069df8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc000069e70, 0xc000021400, 0xfa41f0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:198 +0xcc
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:231
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:388 +0x8b1

```

# Everything's great with 1.8.9, even run against 1.8.10's libraries and config:

```
$ /usr/local/arduino-1.8.9/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999

Ts: 1569351226 - Running: GenerateBuildPathIfMissing
Setting build path to /tmp/arduino-sketch-D41D8CD98F00B204E9800998ECF8427E
Ts: 1569351226 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1569351226 - Running: AddAdditionalEntriesToContext
Ts: 1569351226 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1569351226 - Running: HardwareLoader
Ts: 1569351226 - Running: PlatformKeysRewriteLoader
Ts: 1569351226 - Running: RewriteHardwareKeys
Ts: 1569351226 - Running: ToolsLoader
Ts: 1569351226 - Running: TargetBoardResolver
Ts: 1569351226 - Running: AddBuildBoardPropertyIfMissing
Ts: 1569351226 - Running: LibrariesLoader
Ts: 1569351226 - Running: SketchLoader
Ts: 1569351226 - Running: SetupBuildProperties
Ts: 1569351226 - Running: LoadVIDPIDSpecificProperties
Ts: 1569351226 - Running: SetCustomBuildProperties
Ts: 1569351226 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1569351226 - Running: DumpBuildProperties
_id=leonardo
archive_file_path={build.path}/{archive_file}
bootloader.extended_fuses=0xcb
bootloader.file=caterina/Caterina-Leonardo.hex
```

# 1.8.10 complains that there's no sketch, even if one is defined:

```
$ /usr/local/arduino-1.8.10/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999 Model01-Firmware.ino
stat : no such file or directory
```

# Nightly does the same:

```
$ /usr/local/arduino-nightly/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999 
stat : no such file or directory
```

# #337 PARTIALLY fixes things, but only if given a sketch as an argument:

```

$ /tmp/linux64/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999 Model01-Firmware.ino

/tmp/linux64/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999 Model01-Firmware.ino 
Ts: 1569351539 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1569351539 - Running: AddAdditionalEntriesToContext
Ts: 1569351539 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1569351539 - Running: HardwareLoader
INFO[0000] Loading hardware from: /usr/local/arduino-1.8.10/hardware/ 
INFO[0000] Loading package arduino from: /usr/local/arduino-1.8.10/hardware/arduino 
INFO[0000] Loaded platform                               platform="arduino:avr@1.8.1"
INFO[0000] Excluding directory: /usr/local/arduino-1.8.10/hardware/tools 
Ts: 1569351539 - Running: PlatformKeysRewriteLoader
Ts: 1569351539 - Running: RewriteHardwareKeys
Ts: 1569351539 - Running: TargetBoardResolver
Ts: 1569351539 - Running: ToolsLoader
INFO[0000] Loading tools from bundle dir: /usr/local/arduino-1.8.10/tools-builder/ 
INFO[0000] Loading tools from dir: /usr/local/arduino-1.8.10/tools-builder/ 
INFO[0000] Loaded tool                                   tool=":arduino-preprocessor@0.1.5"
INFO[0000] Loaded tool                                   tool=":ctags@5.8-arduino11"
INFO[0000] Searching tools required for board arduino:avr:leonardo 
INFO[0000] Required tool                                 tool="arduino:arduinoOTA@1.3.0"
INFO[0000] Required tool                                 tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5"
INFO[0000] Required tool                                 tool="arduino:avrdude@6.3.0-arduino17"
Ts: 1569351539 - Running: AddBuildBoardPropertyIfMissing
Ts: 1569351539 - Running: LibrariesLoader
INFO[0000] Adding libraries dir                          dir=/usr/local/arduino-1.8.10/hardware/arduino/avr/libraries location=platform
Ts: 1569351539 - Running: SetupBuildProperties
Ts: 1569351539 - Running: LoadVIDPIDSpecificProperties
Ts: 1569351539 - Running: SetCustomBuildProperties
Ts: 1569351539 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1569351539 - Running: DumpBuildProperties
_id=leonardo
archive_file_path={build.path}/{archive_file}
bootloader.extended_fuses=0xcb
bootloader.file=caterina/Caterina-Leonardo.hex
bootloader.high_fuses=0xd8
bootloader.lock_bits=0x2F
bootloader.low_fuses=0xff
bootloader.tool=avrdude
[...]
````

# With #337 and no sketch as an argument, I'm getting segfaults as the builder tries to abspath an empty string

``` 
/tmp/linux64/arduino-builder -fqbn arduino:avr:leonardo -hardware /usr/local/arduino-1.8.10/hardware/ -tools /usr/local/arduino-1.8.10/tools-builder/ -dump-prefs -warnings all -trace -debug-level 999 
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x84e855]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
	/home/jenkins/go/pkg/mod/github.com/arduino/go-paths-helper@v0.0.0-20190214132331-c3c98d1bf2e1/paths.go:467
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc000095e70, 0xc00016d880, 0xc0001735c0, 0x0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190920124130-b6774033151c/legacy/builder/builder.go:189 +0xf5
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190920124130-b6774033151c/legacy/builder/builder.go:231
main.main()
	/home/jenkins/workspace/arduino-builder-pr-builder/main.go:389 +0x837
```

cmaglie, 2019-09-24:
Thanks for the detailed bug report! I will give a look tomorrow and report back here.

cmaglie, 2019-09-25:
@obra  may you test the build here https://github.com/arduino/arduino-builder/pull/342#issuecomment-535051887 ?

obra, 2019-09-25:
@cmaglie thanks for the super-quick turnaround. That does indeed seem to solve the problem :)

cmaglie, 2019-09-26:
Fixed by #342

closed 2019-09-26 by cmaglie