arduino/arduino-builder #345: When I installed these DHT.h and LiquidCrystal_I2C.h libraries and tried to install program on arduino, I got this error.

More2605, 2019-11-04:
```
goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x130b25a0, 0x13362286, 0x5, 0x13362286)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x130b25a0, 0x132eed60, 0x88ed00, 0x131826f0, 0x13328958, 0x132d3d00, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x130b25a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x130b25a0, 0x1305be1c, 0x20, 0x20, 0x409a01, 0x7fff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x1305bf40, 0x130b25a0, 0xdebbb4, 0x13006c80)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder возвратил 2
Ошибка компиляции для платы Arduino/Genuino Uno.
```

cmaglie, 2019-11-07:
Hi @More2605,

I'm having hard times to reproduce the issue. May you please give me answer these questions to understand what's happening:

- did you install DHT and LiquidCristal_I2C using the library manager? if yes, can you tell me the **exact** library name, I can see many are available. Otherwise can you provide a link to the libraries?
- may you enable "verbose compile" from the preferences, verify the sketch again and post the **full** output? (after enabling verbose compile the output should be much longer and detailed)
- on which operating system are you running on? which board are you using?

cmaglie, 2019-11-07:
* and a copy of the sketch

cmaglie, 2019-11-08:
Fixed by https://github.com/arduino/arduino-cli/pull/477, it will be included in the next release of the builder.

closed 2019-11-08 by cmaglie