arduino/arduino-builder #347: [WIN] Symlinks or junctions not followed in sketch directory

jpmeijers, 2019-12-14:
I have a project consisting of a single folder containing most of my code, and a couple of sketch directories which mostly contain tests and production sketches.

Example layout:
```
Root folder
  + Arduino Libraries
    + OneWire (specific commit pulled in as git submodule)
  + src
    + custom library 1
    + OneWire (symlink to ../Arduino Libraries/OneWire)
  + TestSketchOne
    + TestSketchOne.ino
    + src (symlink to ../src)
```

Inside the sketch I will include the appropriate sources using:
```#include "src/OneWire/OneWire.h"```
```#include "src/custom library 1/library.h"```

This layout works perfectly under Ubuntu, but using Windows it does not. On Windows I get an error saying `testHardware:6:10: error: src/OneWire/OneWire.h: No such file or directory`

Or more verbose:
```
===info ||| Alternatives for src/OneWire/OneWire.h: [] ||| []
===info ||| ResolveLibrary(src/OneWire/OneWire.h) ||| []
===info |||   -> candidates: [] ||| []
C:\Users\jpmeijers\Documents\ebug-v2-arduino-samd21\testHardware\testHardware.ino:6:10: fatal error: src/OneWire/OneWire.h: No such file or directory
 #include "src/OneWire/OneWire.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

It seems like Arduino Builder on Windows does not follow symlinks in the project root directory, neither in the `src` subdirectory - which should be included recursively.

Tested on:
* Arduino IDE v1.8.10
* Arduino IDE v1.9.0-beta (downloaded 20191214)

facchinm, 2019-12-16:
@cmaglie should we move to `arduino-cli` ?

jpmeijers, 2019-12-16:
Using the following configuration symlinks are followed:
1. Clone latest master arduino-cli
2. Clone latest master arduino-builder
3. Add `replace github.com/arduino/arduino-cli => C:\some local dir\arduino-cli` to go.mod.
4. Compile using go sdk 1.13.5

No changes to the code were needed.

per1234, 2021-09-30:
Thanks so much for reporting this @jpmeijers! I am able to reproduce the bug using Arduino IDE 1.8.10 (arduino-builder 1.4.7), but no problem with Arduino IDE 1.8.16 (arduino-builder 1.6.1). So I'll close this as resolved. Feel free to comment if you find the issue still occurs with the latest version.

closed 2021-09-30 by per1234