arduino/arduino-builder #349: Please build MacOS binary using SDK 10.9 or newer

PaulStoffregen, 2020-01-29:
Apple notarization gives a warning that arduino-builder in Arduino 1.8.11 uses a SDK older than 10.9.  Soon Apple will no longer allow notarization for applications with any files failing this check.

To check the SDK version without submitting the entire software package to Apple for notarization, you can run this command:

    otool -l Arduino.app/Contents/Java/arduino-builder | grep sdk

If you run this with Arduino 1.8.11, you'll see arduino-builder was built with SDK 10.7.

cmaglie, 2020-01-29:
Uhm, there is a glich in our CI it seems.

And I think I know where the problem is, working on it...

cmaglie, 2020-02-03:
Here the `arduino-builder` rebuild and signed with the 10.9 SDK:
https://downloads.arduino.cc/tools/arduino-builder-macosx-1.5.1-signed-2.tar.bz2

I'm going to update the download in build.xml of the Arduino IDE accordingly.

closed 2020-02-03 by cmaglie