arduino/arduino-builder #352: Referencing another platform's core with a different name than it's vendorID

PatrickLove, 2020-02-19:
I am trying to define a custom board definition for a PCB I made using at ATMega1281.  I want to mostly extend the [MegaCore](https://github.com/MCUdude/MegaCore) package, but with a few modifications mainly to use a different bootloader since I am running at an unsupported clock rate.

I saw on the [wiki documentation](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool) that it was possible to reference another package's core/variants/platform config, etc.  Among other things I set:

```
boardid.build.core=MegaCore:MCUdude_corefiles
boardid.build.variant=MegaCore:64-pin-avr
boardid.upload.tool=MegaCore:avrdude
boardid.bootloader.tool=MegaCore:avrdude
```

However this fails claiming `MCUdude_corefiles:myvendorid:avr` is missing.  I then found that if I change the core to `boardid.build.core=[anything]:MegaCore` it does find the MegaCore package, but `build.core.path` ends up set to `/path/to/MegaCore/platform/cores/MegaCore` regardless of what I put in the first position.  Since MegaCore has its core folder called MCUdude_corefiles instead, it fails to find any core headers.  The loader in the Arduino IDE complains if the first part isn't a package, but even if you set it to a different package id (like `arduino:MegaCore`) it fails during build trying to include headers (and the log says `Using core 'MegaCore' from platform in folder: /path/to/MegaCore/platform/`).

It looks to me like arduino-builder is using the second half (core name) for both the referenced package and the name of the core to use within the package, and ignores the actual vendorID.  Conveniently this works for the `arduino:arduino` core since they are the same, but it does not seem to work in general.

After going down this rabbit hole, I think I actually can use `arduino:arduino` and only reference the MegaCore variants to get what I was trying to achieve, but this seems like it could cause problems in other situations.

PatrickLove, 2020-02-19:
Never mind, turns out I wasn't using the latest version.  The problem was solved my Arduino install from 1.8.10 to 1.8.12.

closed 2020-02-19 by PatrickLove

matthijskooijman, 2020-02-19:
Thanks for reporting, and the followup when you found out it was fixed :-)

For future reference, this was fixed by https://github.com/arduino/arduino-cli/pull/467