arduino/arduino-builder #359: with_bootloader binary merger failes for large binaries

rikba, 2020-10-02:
Hi community

We are trying to create the binary with bootloader of an arduino sketch using Arduino IDE 1.8.13 and upload it to an ATSAMD21G18A microprocessor (Arduino Zero) via Atmel Studio 7.0.2397. 
It seems to us that the merger that fuses bootloader with sketch code that was fixed in 1.8.13 (https://github.com/arduino/arduino-builder/issues/286, https://github.com/arduino/arduino-cli/pull/744) does not work for larger sketches.

The following two sketches have been uploading via IDE and executed successfully.
However, if the binary is uploaded via Atmel Studio, only the small, first sketch works.

**10992** bytes sketch, works when uploaded with IDE or Atmel as binary.with_bootloader.bin:
```
const char hello[13] PROGMEM = "hello world!";

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  SerialUSB.begin(115200);
}

void loop() {
  SerialUSB.println(hello);
  delay(1000);                       // wait for a second
}
```

**60976** bytes sketch, works when uploaded with IDE, **does not work**  when uploaded via Atmel as binary.with_bootloader.bin:
```
const char hello[50000] PROGMEM = "hello world!";

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  SerialUSB.begin(115200);
}

void loop() {
  SerialUSB.println(hello);
  delay(1000);                       // wait for a second
}
```

Without understanding the merging code, maybe this has to do with some maximum size assumptions https://github.com/arduino/arduino-cli/pull/744/files#diff-b0aa01210cafc5319fbc95159b62da0fR72.

rikba, 2020-10-02:
As a workaround we can upload the bootloader.hex and program.hex successively.

facchinm, 2020-10-05:
@rikba thanks for the report. It looks like the hex merger is maxing out on its own at 65536 bytes (16 bit).
@cmaglie should we add a test in `arduino-cli` for this? The last sketch shows the problem perfectly.

cmaglie, 2020-10-05:
yes this is more an issue for arduino-cli (since the builder is now based on it), I'm moving the issue over there.