arduino/arduino-builder #362: Incorrect placement of auto-generated prototypes gives misleading error messages

6v6gt-duino, 2020-11-01:
This issue is also mentioned here: [https://forum.arduino.cc/index.php?topic=711626.0 (post #7)](https://forum.arduino.cc/index.php?topic=711626.0 (post #7)) 

The expected behaviour is that this simple code example should compile:
`// Arduino IDE 1.8.13

enum class StateA { one, two, three } ;
StateA stateA ;

void setStateA( StateA newState ) {
  stateA = newState ;
}

enum class StateB { one, two, three } ;
StateB stateB ;

void setStateB( StateB newState ) {
  stateB = newState ;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin( 115200 ) ;
  setStateA( StateA::one ) ;
  setStateB( StateB::one ) ;
}

void loop() {
}
`

The observed behaviour is this:

`sketch_oct10a:11:17: error: variable or field 'setStateB' declared void
 void setStateB( StateB newState ) {
                 ^~~~~~
sketch_oct10a:11:17: error: 'StateB' was not declared in this scope
C:\Users\6V6GT\Documents\Arduino\sketch_oct10a\sketch_oct10a.ino:11:17: note: suggested alternative: 'StateA'
 void setStateB( StateB newState ) {
                 ^~~~~~
                 StateA
exit status 1
variable or field 'setStateB' declared void

`

The reason is clearly that one auto-generated function prototype has been wrongly placed (from the file ino.cpp ):
`
#include <Arduino.h>
#line 1 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
enum class StateA { one, two, three } ;
StateA stateA ;

#line 4 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
void setStateA( StateA newState );
#line 11 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
void setStateB( StateB newState );
#line 15 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
void setup();
#line 22 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
void loop();
#line 4 "C:\\Users\\6V6GT\\Documents\\Arduino\\sketch_oct10a\\sketch_oct10a.ino"
void setStateA( StateA newState ) {
  stateA = newState ;
}

enum class StateB { one, two, three } ;
StateB stateB ;

void setStateB( StateB newState ) {
  stateB = newState ;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin( 115200 ) ;
  setStateA( StateA::one ) ;
  setStateB( StateB::one ) ;
}

void loop() {
}
`

facchinm, 2020-11-02:
Hi @6v6gt-duino ,
thank for reporting! The issue lies in the prototype generator as you noticed. What bothers me is that both functions are already declared before their usage so the prototype generator shouldn't create anything.
We'll investigate soon and report back

6v6gt-duino, 2020-12-17:
Here is another link including a discussion of this issue and including additional error/test cases: [https://forum.arduino.cc/index.php?topic=718515.0](url)

sterretjeToo, 2023-07-19:
This topic is now nearly 3 years old. The problem persists in IDE 2.1.1.

> We'll investigate soon and report back

A concerned Arduino user ;)

6v6gt-duino, 2023-07-19:
That gives me the opportunity of adding another example which has occurred in the meantime: https://forum.arduino.cc/t/is-the-location-of-a-global-function-important/996378

6v6gt-duino, 2024-09-02:
Yet another one not seen before from: https://forum.arduino.cc/t/iram-attr-questions/1297715 : 
The _initial_ assessment is that the storage qualifier IRAM_ATTR prevents the identification of second_sharp() as a function and hence no prototype is generated bit this is a guess.
Shifting the function definition above setup() apparently works. 

`#define timepulse D3    //  1 nterrupt every second
volatile int affiche_flag = 0;
volatile int pps_cnt = 0;

void setup() {
  attachInterrupt(timepulse, second_sharp, RISING);
}

void IRAM_ATTR second_sharp() { // routine  d'interruption toute les secondes exactes
  affiche_flag = true; // debloque l'affichage
  pps_cnt = 0;
}

void loop() {
  if (affiche_flag) {
    affiche_flag = false;
    Serial.println("tick");
  }
}`