arduino/arduino-builder #367: arduino-builder returned 255

helo-head, 2021-01-29:
Ive been using the Arduino IDE for about the past two months to program both M5stickc and M5stack Core2 devices. This morning I find that I can no longer compile any of my existing code for any board. I get the same errors each time with no hint as to the issue. To my knowledge there have been no changes to laptop Ive been doing my work on. In an effort to determine the issue I removed the Arduino IDE, reboot and reinstalled the IDE with no change in symptoms. 

The specific error I get is:

signal: killed
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board M5Stack-Core2.

per1234, 2021-09-30:
Hi @helo-head. Thanks so much for reporting this. Are you still having the same issue when using the latest release of Arduino IDE?

helo-head, 2021-09-30:
I had tried removing and reinstalling everything with no luck and as such moved my dev environment to a RBP 4 running Rasbian… Slow but it works …

per1234, 2021-09-30:
Thanks for the quick response. After some research, I believe it is a problem with the ESP32 boards platform dependencies, such as this one:
https://github.com/espressif/arduino-esp32/issues/4408
and not a problem with arduino-builder itself.

That's only a guess, but it will not be possible to achieve any more certainty since you no longer are set up to reproduce the issue. So I'll go ahead and close this, but if anyone has additional information that indicates this is caused by a problem with arduino-builder then you are welcome to comment here and I'll investigate.