arduino/arduino-builder #373: avr-g++: error: device-specs/specs-attiny3226: No such file or directory

haroldmoore, 2021-12-30:
ATtiny3216 an ATtiny 3217 compile up fine but when I try ATtiny3226 or ATtiny3227 I get this :

avr-g++: error: device-specs/specs-attiny3226: No such file or directory 

I used 1.8.13 as required, but still no cigar!  how can I correct this?  is there an easy way?

Thanks in Advance!

Harold

per1234, 2021-12-30:
Hi @haroldmoore. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests for the arduino-builder tool. I don't see any evidence that this is caused by a problem in arduino-builder so I think it will be more appropriate to discuss this further over on the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

You can also take a look at this Arduino Help Center article about an error message in a similar form:
https://support.arduino.cc/hc/en-us/articles/360016119659-Windows-avr-g-error-device-specs-specs-atmega328p-No-such-file-or-directory

closed 2021-12-30 by per1234

haroldmoore, 2021-12-31:
thanks, looks like the fix was a total uninstall of everything, and then
re-install, so far so good,
knock on wood.....

Harold